yorikvanhavre/BIM_Workbench #24: [Question] Import more section types

GreatApo, 2019-06-04:
Thank you for your great work, I am still trying to play around in FreeCAD.

I would like to import sections (UC, UB, CHS, SHS) in the code however I can't find where sections are imported...

yorikvanhavre, 2019-06-05:
Hi! 

It all happens here:
https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Arch/Presets/profiles.csv
Feel free to add new categories! The more we have, the better.
One thing is important to know: the different existing profile types (C, H, R, RH o U) correspond to a specific class in https://github.com/FreeCAD/FreeCAD/blob/master/src/Mod/Arch/ArchProfile.py

If you reuse any of them, there is nothing else to do than adding to profiles.csv.

If you need a new kind of profile, however, you will need 1) to make a new class for it, 2) extend the main makeProfile() function to handle your new profile type. These classes are very simple, you'll see.

If you need further help, consider posting on the [forum](https://forum.freecadweb.org), there is more people there to help!

GreatApo, 2019-06-05:
Awesome, thanks for pointing it out. Hence, the profile types (C, H, R, RH o U) are related to the geometric characteristic of the section in order to create the 3D models. Importing I, H, U and hollow sections is just a matter of copy paste. Then to include angle sections I just need to make a new type (probably copy paste the U :P ).

yorikvanhavre, 2019-06-05:
Coding is 99% copy/pasting anyway :smile:

GreatApo, 2019-06-09:
Good morning,

I added all UB, UC, RHS, CHS sections (https://github.com/GreatApo/FreeCAD/blob/master/src/Mod/Arch/Presets/profiles.csv) and they work fine, however, there seems to be an orientation problem which I don't know if it is related to each section.

[image]

I am following the same place method for all of them: top view -> structure member -> column -> change section -> click on somewhere. HEA sections appear correctly with UC section are placed based on their geometric center with improper length.
Mind that changing the section of the elements based as HEA shows UC correctly.
The same thing happens for other sections too. For example small CHS sections appear properly while large CHS sections are again oriented wrong.

Should I make a pull request with the changes or this is related to the profiles.csv file?
Are there any other changes I should make? I did'n notice sth relative in the ArchProfile.py file...

yorikvanhavre, 2019-06-10:
Everything seems to work correctly for me, using your profiles.csv...
[image]
Are you sure you are using a recent version of FreeCAD? There have been several bugfixes to the structural objects placement recently...

GreatApo, 2019-06-10:
> Everything seems to work correctly for me, using your profiles.csv...
> [image]
> Are you sure you are using a recent version of FreeCAD? There have been several bugfixes to the structural objects placement recently...

Yeah, my mistake, I didn't thought of it. I run the version released in Ubuntu's store. I will see how to compile, test and then pull.

closed 2019-06-10 by GreatApo