yorikvanhavre/BIM_Workbench #27: Can this make a wiring plan?

DieKatzchen, 2019-07-11:
Is it currently possible to design wiring with this workbench? I.e. network cabling, power cabling, video runs, etc.?

yorikvanhavre, 2019-08-09:
You don't really need this workbench to do this, you could do it entirely with Draft tools... However, we could add some more tools to do proper BIM models for these. I think it could be coupled with everything else that needs to follow lines: hydro tubes, ar-co, etc

yorikvanhavre, 2022-02-22:
### Planning

- [ ] Adding other types of sections to Arch Pipes for HVAC: rectangular and slot-shaped
- [ ] Allow to add "flanges" (what is the exact term?)
- [ ] Add default generic hydro appliances
    - [ ]  WC
    - [ ] Faucet
    - [ ] Shower
    - [ ] Bath
    - [ ] Wash machine
    - [ ] Gully
- [ ] Add default generic electrical appliances
   - [ ] Switch
   - [ ] Plug
   - [ ] Light point
   - [ ] Motor
   - [ ] Connection box
- [ ] Add default generic HVAC appliances
   - [ ] Radiator
   - [ ] Exhaust
   - [ ] Intake

TymofiiS, 2023-02-09:
Hi Yorik,
I would like to deeply investigate FreeCAD API for using it for MEP BIM development.
The best way to understand something, as I know,  is participating in its different life aspects.
I would like to help you with this issue, if it appropriate for you.