yorikvanhavre/BIM_Workbench #28: Add version of BIM WB in the WB dropdown menu (like a2plus)

luzpaz, 2019-07-15:
Hey @kbwbe where in the a2plus source can this be found? Thanks!

yorikvanhavre, 2019-07-15:
I don't understand what this issue is about? What are we talking about here??

luzpaz, 2019-07-15:
[image]

yorikvanhavre, 2019-07-15:
Okay, apparently they added the version number in the workbench name. I would advocate against it, because it makes it hard to detect or translate (the day we are able to translate WB names)...

luzpaz, 2019-07-15:
Could we not do (hypothetically) `"translate() + version()"` ?

yorikvanhavre, 2019-07-15:
Theoretically yes, but it's been years that we're discussing WB name translation, so this is unlikely to happen anytime soon

luzpaz, 2019-07-15:
In that case lets just add the version and then when the day arrives when we discuss translation of workbench names then we cross that bridge ? I'm pushing for this because it's very difficult to quickly find the version of installed external workbenches.

luzpaz, 2019-07-15:
Demo on BIM WB: 
[image]

yorikvanhavre, 2019-07-15:
TBH with you I don't like that. I wouldn't like my WB selector to be cluttered with numbers. If you implement something like that, it should be optional.

luzpaz, 2019-07-15:
maybe it can be one of those hidden options ?

yorikvanhavre, 2019-07-15:
What could be done is 1) add a preferences value like "show addon version number". That could be added in one of the pref pages of FreeCAD or in the AddonManager config screen instead. Then 2) instead of the way you used in the PR, check what is the pref value set by the user, and, only if applicable, append the version number like you did.

yorikvanhavre, 2022-02-22:
I think this is becoming obsolete with recent addon manager developments by @chennes

closed 2022-02-22 by yorikvanhavre