yorikvanhavre/BIM_Workbench #37: Error with BIM Properties

ng-galien, 2019-10-22:
Hello,
I'm completely new with FreeCAD and Arch/BIM and Python.
I run Freecad daily on Ubuntu 19.10;
I was trying some tests on it and I have found some errors in the report view when exporting to IFC.
It shows that the IFC property (number of floors) was not correctly parsed.
Then I try to edit IFC properties through the Panel and it gave an error in _BimIfcProperties.py_ File.
I have modified the _updateDicts_ method has showed in the screen shot (sorry for missing pull request), the reference to the property in the dictionary was not correct when adding or modifying it, I don't know about "*VARIES*" case...

`self.objectslist[name][1][prop+";;"+pset]`

[image]

Lines **472**, **474** and **477** in the original file _BimIfcProperties.py_ .

It solved the export problem to, but I'm so far to know if this modification is right for other parts of the app.

Thanks!

yorikvanhavre, 2019-10-22:
Think I got it fixed. Thanks for reporting!

closed 2019-10-22 by yorikvanhavre

ng-galien, 2019-10-22:
Thank you, I have just updated the plugin, it works like a charm ;)

commit 211c39a0 (2019-10-22): Fixed error in properties manager - fixes #37