yorikvanhavre/BIM_Workbench #41: Problems with "Manage IFC quantities"

avicomartin, 2020-04-05:
Hi all,
I am testing BIM module in Freecad and ger errors when use "Manage IFC quantities" option. I tested it in two models. One I made and in a simple new model with only one wall element (Using Freecad 0.18.4 in windows 10 64-bit)

Report view shows:

Running the Python command 'BIM_IfcQuantities' failed:
Traceback (most recent call last):
  File "C:\Users\usuario\AppData\Roaming\FreeCAD\Mod\BIM\BimIfcQuantities.py", line 97, in Activated
    self.update()
  File "C:\Users\usuario\AppData\Roaming\FreeCAD\Mod\BIM\BimIfcQuantities.py", line 114, in update
    self.qmodel.setHorizontalHeaderLabels(translate("BIM",["Label"]) + trqprops)
  File "C:\Program Files\FreeCAD 0.18\Mod\Draft\DraftGui.py", line 65, in translate
    return QtGui.QApplication.translate(context, text, None)

'PySide2.QtCore.QCoreApplication.translate' called with wrong argument types:
  PySide2.QtCore.QCoreApplication.translate(str, list, NoneType)
Supported signatures:
  PySide2.QtCore.QCoreApplication.translate(unicode, unicode, unicode = NULL, int = -1)

luzpaz, 2020-04-05:
Please retest on [0.19_pre](https://github.com/FreeCAD/FreeCAD/releases/tag/0.19_pre), all BIM development is happening in 0.19_pre.

avicomartin, 2020-04-06:
I tried to test 0.19_pre  with this results:

In ubuntu xenial (ppa daily) gets the same error:

Running the Python command 'BIM_IfcQuantities' failed:
Traceback (most recent call last):
  File "/home/antonio/.FreeCAD/Mod/BIM/BimIfcQuantities.py", line 97, in Activated
    self.update()
  File "/home/antonio/.FreeCAD/Mod/BIM/BimIfcQuantities.py", line 114, in update
    self.qmodel.setHorizontalHeaderLabels(translate("BIM",["Label"]) + trqprops)
  File "/usr/share/freecad-daily/Mod/Draft/draftutils/translate.py", line 115, in translate
    return Qtranslate(context, text, None)

'PySide.QtCore.QCoreApplication.translate' called with wrong argument types:
  PySide.QtCore.QCoreApplication.translate(str, list, NoneType)
Supported signatures:
  PySide.QtCore.QCoreApplication.translate(unicode, unicode, unicode = None, PySide.QtCore.QCoreApplication.Encoding = CodecForTr)
  PySide.QtCore.QCoreApplication.translate(unicode, unicode, unicode, PySide.QtCore.QCoreApplication.Encoding, int)

In windows 7 and windows 10, using FreeCAD_0.19.20310-Win-Conda_vc14.x-x86_64.7z downloaded from freecad site I get a dll load error:

  During initialization the error "DLL load failed while importing Part: No se puede encontrar el módulo especificado." occurred in freecad.asm3
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 144, in InitApplications
  File "C:\FreeCAD\bin\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "C:\FreeCAD\bin\lib\site-packages\freecad\asm3\init_gui.py", line 3, in <module>
    from .utils import mainlogger as logger
  File "C:\FreeCAD\bin\lib\site-packages\freecad\asm3\utils.py", line 10, in <module>
    import FreeCAD, FreeCADGui, Part, Draft
ImportError: DLL load failed while importing Part: No se puede encontrar el módulo especificado.
--------------------------------------------------------------------------------

yorikvanhavre, 2020-04-06:
Fixed!

closed 2020-04-06 by yorikvanhavre

luzpaz, 2020-04-06:
Yorik thanks for fixing  
@avicomartin Thanks for reporting. FYI, in the future do you mind putting code within the Mardown triple backtick codeblock for more convenient reading? Thanks

```
This is multiline code
within 2 triple backticks
```

`This is a single line of code within 2 separate backticks`

commit 4c811aea (2020-04-06): Fixed small bug in IfcQuantities - Fixes #41