yorikvanhavre/BIM_Workbench #42: Add user pset

avicomartin, 2020-04-12:
Hi all,
I'm not sure if this is the right place for this question. Sorry if not.

I try to add my own pset to pset_definitions.csv file (I need add particular sets to my objects in all projects: they can arrive to 12 psets with around 150 properties) and for do this I modified the pset_definitions.csv with the format:
PSET_NAME;PSET_PROPERTY_1;PROPERTY_TYPE;PSET_PROPERTY_2;PROPERTY_TYPE; . . . .
for every pset I want to add, but doesn't work. When I try to add pset to my objects in "Manage properties" form, only show psets from IFC schema (psets I added are missing).
Looking the BimIfcProperties.py file, it reads the file pset_definitions.csv to get the properties but previously (if I'm not wrong) it made the pset list from ArchIFCSchema, so my own psets doesn't show.
Is there any way to add user property sets to show when you use "manage properties" form in your objects?

yorikvanhavre, 2020-04-15:
Ok indeed things have changed there with the ArchIFCSchema system. Let me check how to solve it. But indeed it should be allowed to add custom psets like you did. It would even be convenient to do that in a separate .csv file...

closed 2020-04-15 by yorikvanhavre

yorikvanhavre, 2020-04-15:
Ok I don't know what went wrong with your case but what you did should have worked...
Anyway with the above commit there is now a new way: You can now define an additional PSets csv file that will be merged together with the main one. That means you can create your own psets there, or even modify official ones.
The file should be named $USERAPPDIR/BIM/CustomPsets.csv - in the next commit the Ifc Properties dialog now displays an additional text showing the location of that file on your system
It should be placed in

avicomartin, 2020-04-16:
Hi Yorik, thanks for this.

I will test-it.

The way you indicate looks similar to how blender bim has implemented 
custom pset (although so far I haven't managed to get it working 
properly :-) ).

commit 803ff963 (2020-04-15): Allow to define own Psets in a separate csv - fixes #42