yorikvanhavre/BIM_Workbench #45: Issues with BimTutorial on windows

SimonPasquet, 2020-04-26:
Hi !
I get two issues with the bim tutorial on windows 10, freecad 0.18 and the bim workbench updated today (26/04/2020)

1) I get this error when downloading the tutorial
Traceback (most recent call last):
  File "C:\Users\Name\AppData\Roaming\FreeCAD\Mod\BIM\BimTutorial.py", line 122, in load
    f.write(html)
  File "C:\Users\Name\AppData\Local\FreeCAD 0.18\bin\lib\encodings\cp1252.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
UnicodeEncodeError: 'charmap' codec can't encode character '\u200e' in position 4491: character maps to <undefined>

opening offlineloc with "bw" and using "f.write(bytes(html, "utf-8"))" instead of "f.write(html)" resolved the issue. 

2) the images don't load. They are downloaded but it seams that freecad can't find the wright path to them.

luzpaz, 2020-04-26:
Try using FreeCAD 0.19.x and report back

yorikvanhavre, 2020-04-27:
Thanks for the notifications...
I wonder if there isn't a way to write in non-binary mode directly using utf8... Would be better than converting back and forth to bytes. Will try here.
About problem 2), several people reported that as well.. I have no windows install anymore at the moment, so it's hard to find our what's wrong unless somebody helps me to debug... 
There is probably a problem with how the file path is made... Could you try adding this line in BimTutorial.py, at Line165, just above descr=...:
`print ("loading image:",descr.replace(path,"file://"+storename.replace("\\","/")))`
Then when opening the tutorial, you should see a series of image paths printed in the report view.. Could you try copy/pasting one of them into your web browser, see if that works?

yorikvanhavre, 2020-04-27:
Also please check if the above fix works for you?

SimonPasquet, 2020-04-29:
The fix works. I'll try to find what doesn't work for the images next Friday.

yorikvanhavre, 2022-02-22:
This has actually been fixed in da8f2d6fcb0029

closed 2022-02-22 by yorikvanhavre