yorikvanhavre/BIM_Workbench #46: Wall representation

johnnyontheweb, 2020-04-29:
Is it possible to decide which wall representation to be used when exporting/saving IFC?

I need in particular sweptsolid body representation for walls, where to set it?

thanks in advance

luzpaz, 2020-04-30:
Next time, please ask on the [FreeCAD BIM subforum](https://forum.freecadweb.org/viewforum.php?f=23)

johnnyontheweb, 2020-04-30:
By now I cannot reach forum.freecadweb.org

Anyway, I want to make sure that such representation is allowed by code.

luzpaz, 2020-05-01:
> By now I cannot reach forum.freecadweb.org

Try again. The link should work. When you make a thread please link back to this ticket so we can track it.

johnnyontheweb, 2020-05-02:
I don't think I'll have the answer I'm looking for in the FreeCAD Forum, anyway:
https://forum.freecadweb.org/viewtopic.php?f=23&t=45904

yorikvanhavre, 2020-05-04:
At the moment that is not possible, walls are exported as IfcExtrudedAreaSolid when they are straight extrusions, or IfcFacetedBrep in all other cases. Is there any reason in particular why you need swept solids instead of extruded area solids?

johnnyontheweb, 2020-05-04:
Well, please find attached this simple model, I would expect all the wall as IfcExtrudedAreaSolid (even with windows or doors), I found mostly IfcFacetedBrep.
[house.zip](https://github.com/yorikvanhavre/BIM_Workbench/files/4575446/house.zip)

yorikvanhavre, 2020-05-12:
It's very hard to know why without seeing the FCStd file... Might be a question of IFC preferences too (for ex openings are not set to be exported separately), mind to share your FreeCAD file? In any case indeed it's best to do this on the forum

johnnyontheweb, 2020-05-12:
I used for the attached file the option for export openings separately, but same problem.
I find BRep also inside this, despite all walls are straigth. FCstd attached as well.
[simple_house_BIM.zip](https://github.com/yorikvanhavre/BIM_Workbench/files/4615056/simple_house_BIM.zip)

ps.
I'll put a link in the forum to this discussion.

closed 2022-02-15 by johnnyontheweb