yorikvanhavre/BIM_Workbench #49: Error messages: No module found ArchIFC, 'BIMWorkbench' object has no attribute 'draftingtools'

jimmyswimmy, 2020-06-08:
Hi,

I just installed BIM using the instructions on the github readme (used Tools/Addon Mgr) and restarted FreeCAD. The first time I switch to the BIM context, I get the error: "No module named 'ArchIFC'" along with a traceback I usually can't read quickly enough. On subsequent switches back to the BIM context I get the error: "'BIMWorkbench' object has no attribute 'draftingtools'".

Here are my vitals:
```
OS: Windows 10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.4 (GitTag)
Build type: Release
Branch: releases/FreeCAD-0-18
Hash: 980bf9060e28555fecd9e3462f68ca74007b70f8
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

I ran freecad from the CLI with the -l option to obtain the logs and saw the following:
```
Log: Loading Start module... done
Log: Init: Showing main window
Log: Main window restored
Log: Show main window
Log: Toolbars restored
Log: 3Dconnexion device not attached.
Log: Init: Entering event loop
Log: Init: Processing command line files
Log: Module: Part
Log: Loading Part module... done
Err: No module named 'ArchIFC'
Err: Traceback (most recent call last):
  File "<string>", line 102, in Initialize
  File "C:\Users\jim\AppData\Roaming\FreeCAD\Mod\BIM\archguitools\gui_wall.py", line 33, in <module>
    import archmake.make_wall as make_wall
  File "C:\Users\jim\AppData\Roaming\FreeCAD\Mod\BIM\archmake\make_wall.py", line 29, in <module>
    from archobjects.wall import Wall 
  File "C:\Users\jim\AppData\Roaming\FreeCAD\Mod\BIM\archobjects\wall.py", line 35, in <module>
    from ArchIFC import IfcProduct

Log: Loading Arch module... done
Log: Arch workbench activated
Log: Arch workbench deactivated
Err: 'BIMWorkbench' object has no attribute 'draftingtools'
Err: Traceback (most recent call last):
  File "<string>", line 428, in Activated

Wrn: ('[Draft.todo.tasks] Unexpected error:', <class 'Base.FreeCADError'>, 'in ', <built-in function runCommand>, '(', 'BIM_Welcome', ')')Log: Checking for available updates of the BIM workbench
Log: No BIM update available
```

You can see that I selected the BIM module, then went back to Arch, then back to BIM. Any thoughts? I don't see an ArchIFC library anywhere, and I'm not sure about the draftingtools issue.

djacquel, 2020-06-09:
Same for me here.

OS: Ubuntu 18.04.4 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.4.
Build type: Release
Python version: 3.6.8
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

 1. Launching freecad from command line.
 2. New document
 3. Select BIM workbench give this error in the console

> No module named 'ArchIFC'
> Traceback (most recent call last):
>   File "<string>", line 102, in Initialize
>   File "/home/djacquel/.FreeCAD/Mod/BIM/archguitools/gui_wall.py", line 33, in <module>
>     import archmake.make_wall as make_wall
>   File "/home/djacquel/.FreeCAD/Mod/BIM/archmake/make_wall.py", line 29, in <module>
>     from archobjects.wall import Wall 
>   File "/home/djacquel/.FreeCAD/Mod/BIM/archobjects/wall.py", line 35, in <module>
>     from ArchIFC import IfcProduct

  4. Back to any other workbench
  5. Back to BIM workbench gives this error in the console

> 'BIMWorkbench' object has no attribute 'draftingtools'
> Traceback (most recent call last):
>   File "<string>", line 428, in Activated
> 
> ('[Draft.todo.tasks] Unexpected error:', <class 'Base.FreeCADError'>, 'in ', <built-in function runCommand>, '(', 'BIM_Welcome', ')')
>

yorikvanhavre, 2020-06-09:
Thanks for reporting! Indeed the BIM WB should always be kept compatible with 0.18... This should now be fixed with 48a651d

closed 2020-06-09 by yorikvanhavre