yorikvanhavre/BIM_Workbench #50: Error launching Setup: name 'sys' is not defined

VincentFrangi, 2020-06-09:
I occurred an error launching BIM Setup

```
  File "/home/vincent/.FreeCAD/Mod/BIM/BimSetup.py", line 73, in Activated
    if sys.version_info.major < 3:

name 'sys' is not defined
```

OS: Manjaro Linux (i3-with-shmlog)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.21509 (Git)
Build type: Release
Branch: master
Hash: ae641dc5278efafa22b168fc196875558b92e436
Python version: 3.8.3
Qt version: 5.14.2
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)

VincentFrangi, 2020-06-09:
I solve it by importing sys in BimSetup.py file

```python
import sys
```

yorikvanhavre, 2020-06-09:
Fixed in 92619f5
Thanks for notifying!

closed 2020-06-09 by yorikvanhavre