yorikvanhavre/BIM_Workbench #52: Install manually in .19

kb8vsa02, 2020-06-24:


kb8vsa02, 2020-06-24:
has anyone installed BIM in .19 ?? I cannot get FreeCAD to recognize BIM ..

yorikvanhavre, 2020-06-25:
The BIM WB works fine with 0.19. How did you install manually? Probably there was something wrong in the procedure you did...

kb8vsa02, 2020-06-25:
Hi Yorik

Thanks for the reply .. process to install manually that I used was …

1. Download zip file of BIM & Unpak 
2. On Mac .. click on freecad app while holding control key 
3. click on “show package contents”
4. navigate from Contents - Resources - Mods folders 
5. Place unpacked BIM folder in Mods folder 
6. Should show up on Workbench drop down … but does not 

Jeff

mikeprice99, 2020-06-27:
I thought I had the same problem, but then spotted the entry for BIM was at the bottom of the workbench list.

kb8vsa02, 2020-06-27:
Hmm Thanks for the reply Yorik

I checked however and did not see the BIM on worksheet anywhere .

any further thoughts 

Thanks

J

luzpaz, 2020-07-05:
Why not use the [Addon Manager](https://wiki.freecadweb.org/Addon_Manager) ?

yorikvanhavre, 2020-07-06:
@kb8vsa02 It's hard to tell as I don't have a Mac to test... Can you check 1) that there are other workbenches installed in the Mod folder and that these work (otherwise you might simply be using the wrong folder), 2) that your unzipped folder contains the "InitGui.py" file (and several other .py and .ui), and 3) if you open the report panel in FreeCAD, does it give any error message at startup?

kb8vsa02, 2020-07-06:
AM does not seem to be functional either on my Mac with .19 .. don't see anything available on AM

J

kb8vsa02, 2020-07-06:
This is all thats available from AM …

luzpaz, 2020-07-07:
@kb8vsa02 you'll get faster help if you post to the [FreeCAD BIM subforum](https://forum.freecadweb.org/viewforum.php?f=23). Especially since we want to be respectful of Yorik's time (one of the core FC developers). What you're running in to [black screen] is a [Known Issue](https://forum.freecadweb.org/viewtopic.php?t=30573). Open a forum thread and post your [About](https://wiki.freecadweb.org/About) info along with the other data you shared in this ticket. I guarantee you'll get a response.

yorikvanhavre, 2022-02-22:
Closing because of lack of feedback and ways to reproduce. Please reopen if needed!

closed 2022-02-22 by yorikvanhavre