yorikvanhavre/BIM_Workbench #62: Edit IFC properties with Python

fehnomenal, 2020-10-05:
Hi,

first of all, this is a very cool project. I was wondering whether there is a way to programmatically edit IFC properties of components?
Changing the `IfcProperties` dict does not have any effect. I also could not find any place in the code where they would be assigned.

Maybe this is an XY problem?
We want to have a FreeCAD model whose properties (like width, length, ...) can be changed with a python script. Currently this is done with a spreadsheet whose cell values are linked to the geometry.
Now exporting this model to IFC works, but we additionally want to persist the "dynamic" properties as IFC properties.

My vision would be to simply set some `IfcProperties` after setting the spreadsheet values in the python script.
Is there another (perhaps more canonical) way?

yorikvanhavre, 2020-10-05:
Changing the IfcProprties dict does work, yes.
The thing is, ATM you cannot change any of the members of a property directly. You can only replace the property as a whole. ie; this doesn't work:

`myobj.IfcProperties["Length"] = 60`

but this works:

```
d = myobj.IfcProperties
d["Length"] = 60
myobj.IfcProperties = d
```

The IfcProperties dict works a bit specifically, though, I suppose you saw that already, you must write key:value pairs like this:

`"PropertyName;;PropertySet":"PropertyType;;Value"`

example:

`"Description;;WallCommon":"IfcText;;My description text"`

Hope this helps!

fehnomenal, 2020-10-05:
Ok, sorry I just found this thread: https://forum.freecadweb.org/viewtopic.php?f=39&t=46235

Thanks nevertheless.

closed 2020-10-05 by fehnomenal

fehnomenal, 2020-10-06:
Hey Yorik,

one more comprehension question:

If I assign properties the way to described (`'PropertyName;;PropertySet': 'PropertyType;;Value'`) the GUI displays the following:
[image]
Note the "Default property set" string.

If I change it to what seems more naturally to me the dict changes into `'PropertyName': 'PropertySet;;PropertyType;;Value'`:
[image]

The IFC export is exactly the same (modulo `IfcGloballyUniqueId`s). So what exactly is the difference here?