yorikvanhavre/BIM_Workbench #66: AppImage: 0.18 build 16146: 'BIM_Setup' failed: 'BIM_Setup' object has no attribute 'labelVersion'

pawsey-kbuckley, 2020-11-18:
Ubuntu 18.04
Grabbed the AppImage
Made it executable
Ran it
Installed the BIM workbench from the Addon manager
Clicked on BIM Setup menu item under the manage menu

Saw an error message in the statusbar.

On the console from which the AppImage had been run, 
saw the following

```
$ ./FreeCAD_0.18-16146-rev1-Linux-Conda_Py3Qt5_glibc2.12-x86_64.AppImage 
/tmp/.mount_FreeCAFUfxJn
FreeCAD 0.18, Libs: 0.18R16146 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2019
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Unknown command 'Draft_Arc_3Points'
Unknown command 'Draft_Arc_3Points'
Running the Python command 'BIM_Setup' failed:
Traceback (most recent call last):
  File "/home/kbuckley/.FreeCAD/Mod/BIM/BimSetup.py", line 121, in Activated
    t = self.labelVersion.text

'BIM_Setup' object has no attribute 'labelVersion'
```

Any clues?

luzpaz, 2020-11-18:
Test with v0.19 and report back

pawsey-kbuckley, 2020-11-18:
Now see

```
$ ./FreeCAD_0.19-22960-Linux-Conda_glibc2.12-x86_64.AppImage
FreeCAD 0.19, Libs: 0.19R22960 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2020
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

During initialization the error "'ascii' codec can't decode byte 0xc3 in position 210: ordinal not in range(128)" occurred in /tmp/.mount_FreeCA5AQ4Wn/usr/Mod/Surface/InitGui.py
Please look into the log file for further information
```

however, the BIM workbench does now get setup for use OK 
and running the 

```
import ifcopenshell
```

in the Python console, returns without error.

Cheers,
Kevin

luzpaz, 2020-11-18:
That issue is unrelated. You can close this ticket as we won't be backporting fixes for 0.18.x

closed 2020-11-18 by pawsey-kbuckley