yorikvanhavre/BIM_Workbench #67: Error when running BIM Setup on FreeCAD 0.19.23074

russss, 2020-11-29:
It seems like DraftToolBar has changed in the dev version of FreeCAD, and setup fails to run with the error `'DraftToolBar' object has no attribute 'widthButton'`. The `fontsizeButton` member also no longer exists.

I'm not currently familiar enough with FreeCAD's scripting to work out what to replace these with, but commenting [this block](https://github.com/yorikvanhavre/BIM_Workbench/blob/f2f4057f439d19526d03553db6c7ee65b610cd3d/BimSetup.py#L211-L213) out allows setup to run successfully.

yorikvanhavre, 2020-12-02:
Thanks for notifying! Fixed!

closed 2020-12-02 by yorikvanhavre

freecadcostarica, 2021-01-23:
Just downloaded the Bim from Github and  i cannot setup my proyect, seems that is the  BIM_setup.py thanks for your work!
[image]

yorikvanhavre, 2021-03-01:
Further fix in 6d8c87c .. This should solve the above error

commit 8d145a02 (2020-12-02): Fixed BimSetup - fixes #67