yorikvanhavre/BIM_Workbench #75: BIM Views Manager Icons have a fixed "maximumSize" width. Width above 32 px Non Styleable.

turn211, 2021-03-03:
As per discussion https://forum.freecadweb.org/viewtopic.php?f=34&t=56219 maybe when time is available some solutions. Thanks
[image]

yorikvanhavre, 2022-02-22:
Sorry it took so long to look at this :sweat_smile: I am not sure how to specify a max width (in case no stylesheet is used), but I'll try removing the maxwidth property and see what happens

yorikvanhavre, 2022-02-22:
Hmm if I do that I can't make a narrow dialog anymore... Maybe just removing the spacer would do?

turn211, 2022-02-22:
I think you have to change the spacers from having a Maximum size to Expanding. 
[https://programmerall.com/article/5185347729/](url)