yorikvanhavre/BIM_Workbench #78: BIM_Workbench initialization error

tjleon1, 2021-07-17:
I installed FreeCAD, activated a couple of other workbenches (fasteners, sheetmetal and parts_library with restarts each time), then BIM_Workbench and a restart.  When I selected the workbench, I was presented with the following error:

13:20:13  Traceback (most recent call last):
13:20:13    File "C:\Users\Tim\AppData\Roaming\FreeCAD\Mod\BIM\BimSetup.py", line 413, in setPreset
13:20:13      if colSimple != None:
13:20:13  UnboundLocalError: local variable 'colSimple' referenced before assignment

Probably a simple fix, but I'm unfamiliar with the project and with python in general.

luzpaz, 2021-07-17:
Please always add your FreeCAD [About](https://wiki.freecadweb.org/About) info when reporting issues. thanks

allanlaal, 2021-07-18:
OS: Ubuntu 21.04 (MATE/mate)
	Word size of FreeCAD: 64-bit
	Version: 0.20.25220 (Git) AppImage
	Build type: Release
	Branch: master
	Hash: 46282db7c8c65d1205a4cd03499d4beadb1573c6
	Python version: 3.9.6
	Qt version: 5.12.9
	Coin version: 4.0.0
	OCC version: 7.5.2
	Locale: English/United States (en_US)

	Traceback (most recent call last):
	  File "/home/allan/.FreeCAD/Mod/BIM/BimSetup.py", line 413, in setPreset
		if colSimple != None:
	UnboundLocalError: local variable 'colSimple' referenced before assignment

luzpaz, 2021-07-18:
Possibly something to do with @yorikvanhavre's commit 89469eba00bcb30f06fed153a622723b26f0b26b ?

luzpaz, 2021-07-19:
@allanlaal and @tjleon1 

Please test the fix by doing the following (assuming you're on linux)
```
cd ~/.FreeCAD/Mod/BIM/
git fetch 
curl -ss https://patch-diff.githubusercontent.com/raw/yorikvanhavre/BIM_Workbench/pull/80.patch | git apply
```
Restart FreeCAD  
Try to recreate the bug

allanlaal, 2021-07-19:
@luzpaz that fixed it! :)

commit 52840f15 (2021-07-27): Fixed initialization error - fixes #78

Fixes https://github.com/yorikvanhavre/BIM_Workbench/issues/78
closed 2021-07-27 by yorikvanhavre