yorikvanhavre/BIM_Workbench #81: Unable to generate custom properties than are exported to IFC

bje-, 2021-09-29:
I am trying to add some custom properties to objects in FreeCAD so that they appear in the exported IFC file. Do these need to be added in a particular way for this to happen?

yorikvanhavre, 2021-12-01:
Sorry, I ended up overlooking this :grimacing: 
There is currently no way to do this apart from using the IFC properties manager. There, you can create an IFC property set and adding custom properties to it.

But there is some time I think we should switch to a full FreeCAD properties system. That is, all the IFC properties appear in the FreeCAD object properties, and vice-versa (all FreeCAD object properties are saved to IFC).

This is a good opportunity to define how to do this. The problems I see:

* Not all FreeCAD properties must be exported to IFC. Some are internal to FreeCAD.
* Not all FreeCAD properties CAN be exported to IFC. Ex: Placements
* Some IFC properties might need to be mapped to existing, default FreeCAD properties. Ex, name

We could do this in different steps:

1. Keep everything as it is, and additionally export some FreeCAD custom properties as IFC properties
2. Find a system to deal with different or unsupported property types 
3. Migrate the current IfcProperties dict to FreeCAD properties
4. Redesign the IFC Properties manager

bje-, 2021-12-02:
Sounds good. I am not a FreeCAD developer, I just want something That Works. :-)

yorikvanhavre, 2022-02-22:
### Redesign plan

Sync between IFC properties and FreeCAD properties

#### In Arch:

- [ ] IFC properties are stored as FreeCAD properties
- [ ] Property sets are stored as FreeCAD property groups
- [ ] Need a conversion map between FreeCAD property types and IFC property types
- [ ] IFC Importer: map to FreeCAD properties
- [ ] IFC exporter: map from FreeCAD properties

#### In BIM:

- [ ] Redesign of the properties manager UI: Allow to add/remove properties (sync with FreeCAD properties), mark groups as exported to IFC or not, import property sets

yorikvanhavre, 2022-09-15:
First test in [123440c](https://github.com/yorikvanhavre/BIM_Workbench/commit/123440c88214e0d10a84de4c7181cf28dfd9cff5)