yorikvanhavre/BIM_Workbench #84: Test issue for attachments

yorikvanhavre, 2021-12-14:
This is a dummy, test issue to test
1) If we can attach .FCStd files
2) If attachments can be downloaded locally or via the API

yorikvanhavre, 2021-12-14:
Ok point 1. failed, FCStd files are not permitted

Trying point 2 with an attached image:
[image]

yorikvanhavre, 2021-12-14:
Above image is saved on a github website and has a public link so it can be downloaded.
Trying with a zip file..
[test.FCStd.zip](https://github.com/yorikvanhavre/BIM_Workbench/files/7711484/test.FCStd.zip)
.

yorikvanhavre, 2021-12-14:
This is a comment posted from ghi

luzpaz, 2021-12-15:
@yorikvanhavre maybe we can ask github to allow FCStd files to be supported?

yorikvanhavre, 2021-12-15:
@luzpaz apparently custom filetypes are a paid feature of github, so it's very unlikely they will give us for free... @donovaly has had a look already, but getting ourselves a paid github account seems also unpractical (you need to get one for each collaborator) :roll_eyes:

donovaly, 2021-12-15:
> @luzpaz apparently custom filetypes are a paid feature of github,

I thought I read about this. Yesterday I had a closer look and it turned out that this is not the case.

> but getting ourselves a paid github account seems also unpractical (you need to get one for each collaborator) 🙄

Reading the GitHub plans:
https://github.com/pricing
it seems so. Then I asked my IT (we use GitLab but it offers more or less the same plans as GitHub) and as I understood it, every "user" does not mean every contributor, but person who can create releases, merges etc.
I also researched a bit in forums and as we have seems already the optimum for projects like us.
However, I might have made mistakes or misunderstood something.

yorikvanhavre, 2022-02-22:
I'll close this issue because it was just a test issue.
Regarding .FCStd support there is probably little that can be done while we are on github, but let's keep an eye open.

closed 2022-02-22 by yorikvanhavre