yorikvanhavre/BIM_Workbench #85: Status Bar didn't show in case of default WB is BIM

ebrahimraeyat, 2022-01-05:
first, Thank you so much for this great WB.

When I set the BIM workbench as default WB, instead of appear in status bar, the update icon, unit widget and other appears as a separate toolbar.

In my WB, I implemented this method, but for mine, it didn't show anything, I mean It did not show even separate toolbar and when program search for statusbar ([in your code that i borrowed](https://github.com/yorikvanhavre/BIM_Workbench/blob/c6b381044879af7a502279248dc7fdeb8d6f8144/BimStatusBar.py#L164)), it return None. But when I set another WB as default it appears.

[in my code](https://github.com/ebrahimraeyat/civilTools/blob/7c75ce3da23ee1331ec396376b4f3c28acf04c56/CivilToolsStatusBar.py#L84)

yorikvanhavre, 2022-01-11:
Hi,
Indeed. The status bar system was implemented in Draft only so far.
Do you find it better in the status bar? I kind of like having it in the toolbars, because I usually disable the status bar. Maybe we could allow both?

ebrahimraeyat, 2022-01-11:
Hello @yorikvanhavre .
this way, maybe user did not found it. because if i set for example BIM or My workbench as default WB in FreeCAD, instead of appears in status bar, it appears in toolbar and maybe confusing user. Thanks

yorikvanhavre, 2022-02-22:
I think @realthunder's branch allows to add toolbars to the status bar, so this could become obsolete (user decides where they place their toolbars...)