yorikvanhavre/BIM_Workbench #89: 'float' object cannot be interpreted as an integer when starting workbench

TheSimpleZ, 2022-02-10:
Hello! 

Running the BIM workbench after a fresh install of FreeCAD gives me this error:

```sh
23:58:02  Running the Python command 'BIM_Views' failed:
Traceback (most recent call last):
  File "~/.FreeCAD/Mod/BIM/BimViews.py", line 119, in Activated
    vm.setFloating(pref.GetFloat("ViewManagerFloating",False))

'float' object cannot be interpreted as an integer
```

Any ideas what could be causing it?

luzpaz, 2022-02-11:
Please always post your full [About](https://wiki.freecad.org/About) info of FreeCAD. Are you using the latest BIM workbench?

luzpaz, 2022-02-11:
Can't reproduce on
```
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.27472 (Git)
Build type: Release
Branch: master
Hash: ba42231b9c6889b89e064d6d563448ed81e376ec
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.0
OCC version: 7.5.2
Locale: English/United States (en_US)
```
Running 'latest' BIM workbench (although not sure what version because addon manager doesn't show (https://github.com/FreeCAD/FreeCAD-addons/issues/199)

luzpaz, 2022-02-11:
@TheSimpleZ can you test it on the latest v0.20  
build https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

TheSimpleZ, 2022-02-11:
Hmm... I'm having trouble using that build. But it seems related to Freecad itself rather than the BIM workbench. So I uninstalled Freecad and got freecad-git from AUR.  However, when I open it, I just get a blank screen with an error message saying `21:57:51  No module named 'StartGui'`. I haven't managed to figure out what's wrong yet. 

```
OS: Manjaro Linux
Word size of FreeCAD: 64-bit
Version: 0.20.27489 (Git)
Build type: Release
Branch: master
Hash: 09a05a9cd0c4692a57a3e038268b4389b4657fc6
Python version: 3.10.2
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/Denmark (en_DK)
```

[image]

Once I get it working I'll report back. If you have any tips on how to fix this, it would be appreciated :)

luzpaz, 2022-02-11:
Can you try using the [AppImage](https://wiki.freefcad.org/AppImage) or the [Snap package](https://wiki.freefcad.org/Snap) ?

TheSimpleZ, 2022-02-11:
Hmm... Quite strange but neither of them works... 

I tried running the AppImage. It quickly fails with 
```
➜  ~ Downloads/FreeCAD_weekly-builds-27319-Linux-Conda_glibc2.12-x86_64.AppImage 
FreeCAD 0.20, Libs: 0.20R27319 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

User path has changed to /home/zrean/.local/share/FreeCAD/. Please move user modules and macros
/tmp/.mount_FreeCAhKHIXZ/AppRun: line 41: 1051035 Aborted                 (core dumped) ${MAIN} "$@"
```

Then I installed the snap package using `snap install --edge freecad-ppd`
Now I managed to run version 0.20 of FreeCAD. However, as soon as I try to open any document, even a new one, FreeCAD crashes. This is even before I install the BIM workbench. 

I tried running the snap from the terminal and got this crash message when creating a new document:
```
➜  ~ freecad-ppd.freecad
FreeCAD 0.20, Libs: 0.20R27472 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Adding snap-specific PYTHONPATH to sys.path: /home/zrean/snap/freecad-ppd/common/.local/lib/python3.8/site-packages:/snap/freecad-ppd/154/lib/python3.8/site-packages:/snap/freecad-ppd/154/usr/lib/python3/dist-packages
connect failed: No such file or directory
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f6642875210]
#1  /snap/freecad-ppd/154/kf5/usr/lib/x86_64-linux-gnu/libX11.so.6(XDefaultScreenOfDisplay+0x4) [0x7f664062cbc4]
#2  /snap/freecad-ppd/154/usr/lib/x86_64-linux-gnu/libCoin.so.80c(+0x3eeb5e) [0x7f66423d9b5e]
#3  /snap/freecad-ppd/154/usr/lib/x86_64-linux-gnu/libCoin.so.80c(glxglue_init+0x40) [0x7f66423da470]
#4  /snap/freecad-ppd/154/usr/lib/x86_64-linux-gnu/libCoin.so.80c(cc_glglue_instance+0x235) [0x7f66423d2fb5]
#5  0x7f66422751a8 in SoGLRenderActionP::isDirectRendering(SoState const*) const from /snap/freecad-ppd/154/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0x48
#6  0x7f6642277577 in SoGLRenderActionP::render(SoNode*) from /snap/freecad-ppd/154/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0xc7
#7  0x7f66422777c3 in SoGLRenderAction::beginTraversal(SoNode*) from /snap/freecad-ppd/154/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0xa3
#8  0x7f664227071a in SoAction::apply(SoNode*) from /snap/freecad-ppd/154/usr/lib/x86_64-linux-gnu/libCoin.so.80c+0x28a
#9  0x7f664555737d in Gui::SoBoxSelectionRenderAction::apply(SoNode*) from /snap/freecad-ppd/154/usr/lib/libFreeCADGui.so+0x2d
#10  0x7f66456119a5 in Gui::View3DInventorViewer::renderScene() from /snap/freecad-ppd/154/usr/lib/libFreeCADGui.so+0x145
#11  0x7f66455d2fed in SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*) from /snap/freecad-ppd/154/usr/lib/libFreeCADGui.so+0x30d
#12  0x7f66455dadd6 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::paintEvent(QPaintEvent*) from /snap/freecad-ppd/154/usr/lib/libFreeCADGui.so+0x46
#13  0x7f6643ab5cf6 in QWidget::event(QEvent*) from /snap/freecad-ppd/154/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x286
#14  0x7f6643b660e2 in QFrame::event(QEvent*) from /snap/freecad-ppd/154/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x22
#15  0x7f6643dd9501 in QGraphicsView::viewportEvent(QEvent*) from /snap/freecad-ppd/154/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x161
#16  0x7f66455d30ce in SIM::Coin3D::Quarter::QuarterWidget::viewportEvent(QEvent*) from /snap/freecad-ppd/154/usr/lib/libFreeCADGui.so+0x9e
#17  0x7f6642ef5b13 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /snap/freecad-ppd/154/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0xa3
#18  0x7f6643a72db2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad-ppd/154/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x72
#19  0x7f6643a7bbb8 in QApplication::notify(QObject*, QEvent*) from /snap/freecad-ppd/154/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1a8
#20  0x7f66452b1e48 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad-ppd/154/usr/lib/libFreeCADGui.so+0x88
#21  0x7f6642ef5daa in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad-ppd/154/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12a
#22  0x7f6643aae0aa in QWidgetPrivate::sendPaintEvent(QRegion const&) from /snap/freecad-ppd/154/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x3a
#23  0x7f6643aae8d2 in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, QFlags<QWidgetPrivate::DrawWidgetFlag>, QPainter*, QWidgetRepaintManager*) from /snap/freecad-ppd/154/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x7d2
[... cut ...]
#75  0x7f6642ef48ab in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /snap/freecad-ppd/154/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#76  0x7f6642efca64 in QCoreApplication::exec() from /snap/freecad-ppd/154/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#77  0x7f664521965a in Gui::Application::runApplication() from /snap/freecad-ppd/154/usr/lib/libFreeCADGui.so+0x179a
#78  /snap/freecad-ppd/154/usr/bin/FreeCAD(+0x4eb0) [0x55a906d1ceb0]
#79  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f66428560b3]
#80  /snap/freecad-ppd/154/usr/bin/FreeCAD(+0x524e) [0x55a906d1d24e]
```

TheSimpleZ, 2022-02-12:
I tried running the stable AppImage, and I got the same error. I now tried running the stable version of the snap and it seems to work fine. No errors, I can load my file and the BIM workbench seems to work fine. 

I'm puzzled by why the new version crashes, but for now, I'm able to do what I need. Thanks for the support 😄

luzpaz, 2022-02-12:
jeez, wth is going on lol

yorikvanhavre, 2022-02-22:
Thanks for notifying, both bugs should be fixed now!

commit 3a12c1f5 (2022-02-22): Fixed small bugs - fixes #89
closed 2022-02-22 by yorikvanhavre