yorikvanhavre/BIM_Workbench #92: Better support for IfcAnnotations

yorikvanhavre, 2022-03-23:
Have a solid, round-trippable support for 2D annotations (lines, texts, dimensions,...) in IFC, through [IfcAnnotations](https://standards.buildingsmart.org/IFC/DEV/IFC4_2/FINAL/HTML/schema/ifcproductextension/lexical/ifcannotation.htm).

Related issue @ BlenderBIM: https://github.com/IfcOpenShell/IfcOpenShell/pull/2100

* [ ] Support [Draft Labels](https://wiki.freecadweb.org/Draft_Label)
* [ ] Set ObjectType when exporting an IfcAnnotation. In FreeCAD I suppose we would have:
   * Dimensions
   * Linework
   * Texts
   * Areas/surfaces (faces and hatches)
   * Smart labels
   * Undefined (for all the rest I cannot think of)
* [ ] For "usermade" labels, (a line + a text), find a way to export them as one single IfcAnnotation
* [ ] Add IfcRelAssignsToProduct when exporting smart labels and dimensions (if associated to an object)
* [ ] associate it with a drawing object in a group, or within a drawing (e.g. IfcDocumentInformation)

Moult, 2022-03-24:
Cheers, most of these already have solutions worked on as part of the IfcOpenShell drawing module, so let's perhaps start with the easiest then work our way up. If you're curious on reading this is a good post to read: https://github.com/IfcOpenShell/IfcOpenShell/issues/1153#issuecomment-749491531

So to start with:

 1. Camera object will be stored in IFC as an IfcAnnotation, it must have an ObjectType of DRAWING. (Camera is Blender jargon, but it's basically the view foreground cutting plane + background cutting plane extents)
 2. IfcAnnotation[ObjectType="DRAWING"] must belong in a single IfcGroup
 3. The DRAWING annotation object must store an IfcBlock in the MODEL/Body/MODEL_VIEW subcontext.
 4. The DRAWING annotation object ObjectPlacement should have a rotation such that -Z points at the direction you are looking at in the section cut (i.e. same as BCF), +X points to the right, and +Y points up the drawing.
 5. The DRAWING annotation object should store a Pset called EPset_Drawing
EPset_Drawing should contain: TargetView (enum to match subcontext TargetView attribute), Scale (IfcLabel, like "X/Y" where X and Y are integers)

Once you have these, that should be enough to get a drawing translated across. Hopefully the logic why we need each of these data should be obvious but if it isn't let me know and happy to have a call to explain the reasoning.

yorikvanhavre, 2022-03-24:
Thanks for the detailed explanation. Okay, I think that's pretty easy to implement. Two questions:

* Is the IfcBlock the shape representation of the IfcAnnotation/DRAWING? Then it's basically a "view cube"? That's handy, we have that data in FreeCAD already.
* I never saw EPset prefix before... That's elegant, to differentiate from standard psets. Is that a standard? I can't find any reference...

Moult, 2022-03-24:
Exactly, the IfcBlock is the "view cube". Everybody (I assume) needs a view cube to represent the full 3D drawing extents so I think it works well and is agnostic of camera properties.

"EPset_" is a little hack where something is a custom pset, but I consider it to be an extension of IFC itself or a workaround that is common to many implementations. It was an unofficial convention stablished by the georeferencing workarounds in IFC2X3 when IFC2x3 needed EPset_ProjectedCRS and EPset_MapConversion.

yorikvanhavre, 2022-03-30:
[image]
I think we could achieve something good here, being able to design such things in either FreeCAD or Blender and render it in the other... I haven't looked deep into finer text styling, if things like font, alignment... are possible

Moult, 2022-03-30:
By the way I always use IfcTextLiteralWithExtents because that gives me the option of using a BoxAlignment. The BlenderBIM Add-on has a tiny Style manager which includes all the 4 presentations styles (surface, curve, fillarea, and text).

Right now though the styles are done with CSS in SVG, however I do plan to let the IFC styles generate the CSS so they are linked.

yorikvanhavre, 2022-03-31:
i like css and svg too.. the thing is how to transport drawings between, say 😇 blender and freecad, in a manner that they don't loose important things like text size and position. For now indeed having that in the ifc file is simple. but indeed, think of css-stylable cad drawings!

yorikvanhavre, 2022-03-31:
oops, closed by mistake..

yorikvanhavre, 2022-04-06:
As of https://github.com/FreeCAD/FreeCAD/commit/b5a6970c5317f3b82788ea9646d3e80e356b36d5 IfcAnnotations exported from FreeCAD now have these possible ObjectType: LINEWORK,AREA,HATCH,LEADER,TEXT,DIMENSION, or None (empty string)

Moult, 2022-04-15:
A sample file that includes some sheet and drawing info :)

[untitled.ifc.txt](https://github.com/yorikvanhavre/BIM_Workbench/files/8498801/untitled.ifc.txt)

theoryshaw, 2022-06-15:
Related: A thread on how to codify dimension offsets in the IFC schema: https://community.osarch.org/discussion/1014/ifc-an-approach-for-dimension-line-offsets/

theoryshaw, 2022-09-07:
To help fund this issue directly, OSarch created a 'bounty' here: https://opencollective.com/osarch/projects/roundtrip_ifc_annotation