yorikvanhavre/BIM_Workbench #93: Failling to load draft tools

VincentFrangi, 2022-04-01:
Hello

I'm occurring this issue when I open the BIM workbench:

I don't have the draft tools in the tool bar

```
11:29:51  Traceback (most recent call last):
  File "<string>", line 453, in Activated

11:29:51  'BIMWorkbench' object has no attribute 'draftingtools'
```

I got [freecad-linkstage3-bin](https://aur.archlinux.org/packages/freecad-linkstage3-bin) package from AUR
```
OS: Manjaro Linux (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.20.Unknown
Build type: Release
Python version: 3.10.2
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```

Thank you in advance
Vincent

VincentFrangi, 2022-04-06:
Now it's fine, not sur why but I reinstall with overwrite files of [med package](https://archlinux.org/packages/community/x86_64/med/)
```
yay -S med --overwrite '*'
```

yorikvanhavre, 2022-04-07:
Ok, no idea why med should have fiddled with the BIM workbench, but glad you have it working!