yorikvanhavre/BIM_Workbench #94: IFC-compliant workflow

yorikvanhavre, 2022-06-19:
Idea: being able to load and save IFC files with minimal or no data loss. while preserving FreeCAD's free modelling paradigms that are not too tied to the IFC structure

### Workflow

* Open an IFC file
* Modify some elements
* Save the IFC file
* Only the modified elements are modified, the rest stays the same

### Implementation

* The file is imported "the normal way", this is different than BlenderBIM: not supported entities are discarded, and the different import styles of FreeCAD continue to work
* When opening an IFC file, an entry is created in the document meta, referring to the original IFC file, wtih sha hash
* Alternative: The whole IFC file can be embedded?
* IFC objects are created with an "unmodified" tag
* The user modifies, creates objects, etc... normally. For each of them, the "unmodified" tag is removed. Deleted objects are added to some delete log
* When saving to IFC (being the original IFC file or a new one), the original file is read first
* If  IFC data is unavailable, or if the sha has changed, the user is warned
* The ifcbin is ran on the file to collect/catalog existing elements
* The existing building structure is collected
* New IFC shape representations are generated
* Existing objects are updated, new objects are created, deleted objects are deleted. Attached shape representation sare collected
* Shape representations are deep-deleted, with elements that only depend on them
* Exported objects are marked as unmodified
* The new hash is stored, or the included IFC content is updated

brunopostle, 2022-07-04:
Linking to this discussion on 'Native IFC' definitions for reference: IfcOpenShell/IfcOpenShell#2240

luzpaz, 2022-07-04:
Sweet, thanks @brunopostle !

yorikvanhavre, 2022-09-15:
To use the new BIM IFC importer where most of the work here is gonna be done, several options:

1) Use latest FreeCAD code, then there is nothing to do (just make sure Preferences -> import/export -> IFC -> multicore is > 0)
2) In Open/Import file dialog, make sure to select "Industry Foundation Classes (BIMWB)" as the importer

Moult, 2022-10-15:
I might recommend starting with a smaller scope, like attribute editing :) I'm super keen for this to be a reality!