yorikvanhavre/BIM_Workbench #95: Add on manager has disfunctional link

liftoff-sr, 2022-06-20:
Ubuntu x86_64 20.04, 
FreeCAD 0.19

I am unable to install the workbench, the URL is disfunctional.  I tried with wget also:

This:
wget https://github.com/yorikvanhavre/BIM_Workbench/archive/master.zip
fails too.

luzpaz, 2022-06-21:
Link works fine for me with `wget`:  
`2022-06-20 20:46:30 (1.00 MB/s) - ‘master.zip’ saved [3497094]`

luzpaz, 2022-06-21:
@yorikvanhavre probably good to close ticket. OP can re-open if necessary

closed 2022-06-21 by yorikvanhavre

luzpaz, 2022-06-21:
The URL for codeload works for me. What country are you downloading from ?

liftoff-sr, 2022-06-22:
$ ping codeload.github.com

this hangs as well. My internet service provider is T-Mobile, a 5G cellular modem product from T-Mobile for business users.

My public IP would be dynamically allocated from a block in this range:

172.32.0.0/11

Last attempt was from 172.56.88.5 and this failed the wget and the ping both.

luzpaz, 2022-06-22:
No worried :stuck_out_tongue_winking_eye: