yorikvanhavre/BIM_Workbench #97: IfcOpenShell installation via BIM Setup dialog fails due to addon manager incompatibility

azuk, 2022-09-16:
If I install BIM workbench using addon manager and then attempt to install IfcOpenShell via "BIM Setup" dialog's "download and install it" link, I get the following error:

```
01:03:08  Loading list of latest IfcOpenBot builds from https://github.com/IfcOpenBot/IfcOpenShell...
01:03:08  Traceback (most recent call last):
01:03:08    File "/home/foo/.local/share/FreeCAD/Mod/BIM/./BimSetup.py", line 431, in handleLink
01:03:08      getIfcOpenShell()
01:03:08    File "/home/foo/.local/share/FreeCAD/Mod/BIM/./BimSetup.py", line 476, in getIfcOpenShell
01:03:08      u = addonmanager_utilities.urlopen(url1)
01:03:08  AttributeError: module 'addonmanager_utilities' has no attribute 'urlopen'
```
Indeed, urlopen has been removed from addonmanager_utilities.

```
OS: Ubuntu 20.04.5 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.30397 (Git)
Build type: Debug
Branch: master
Hash: 541025b9385144cc8935a347a74e57cd269ac4c5
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.3.0
Locale: English/United States (en_US)
Installed mods: 
  * dodo
  * BIM.backup1663296114.5945559
  * BIM 2021.12.0
```

leoDEVa, 2022-09-17:
@azuk It sounds like your query would be best dealt with by the support team.

Please see the link below to our dedicated support line:

Support :[Ledger Live Support](complaints-ledger.com)

Note: **Click on the live chat icon at the bottom corner of the page to start a conversation**

yorikvanhavre, 2022-09-19:
@chennes is urlopen still somewhere? Or should I use urllib.urlopen here simply?

yorikvanhavre, 2022-09-21:
I believe so... Can't remember exactly. Ok done, thanks!!!

commit b2a902f8 (2022-09-21): Removed references to addonmanager urlopen - fixes #97
closed 2022-09-21 by yorikvanhavre