yorikvanhavre/BIM_Workbench #98: Syntax error in latest commit...

swingwinger, 2022-09-22:
I am receiving the following error when attempting to initialize the latest BIM_Workbench commit in Freecad 0.20.1

00:32:20  invalid syntax (BimSetup.py, line 464)
00:32:20  Traceback (most recent call last):
  File "<string>", line 84, in Initialize
  File "/tmp/.mount_FreeCArviCMj/usr/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)

luzpaz, 2022-09-22:
Please always add you full [About](https://wiki.freecad.org/About) info.

swingwinger, 2022-09-22:
Here is the about info...

[code]
OS: Linux Mint 19.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.1)
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * BIM 2021.12.0
[/code]

swingwinger, 2022-09-22:
Similarly under Windows, this error:

09:55:21  invalid syntax (BimSetup.py, line 464)
09:55:21  Traceback (most recent call last):
  File "<string>", line 84, in Initialize
  File "C:\Program Files\FreeCAD 0.20\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * BIM 2021.12.0
[/code]

htlcnn, 2022-09-22:
it is obvious: `invalid syntax (BimSetup.py, line 464)`

fernom, 2022-09-22:
Same error:
```
[code]
13:34:21  invalid syntax (BimSetup.py, line 464)
13:34:21  Traceback (most recent call last):
  File "<string>", line 84, in Initialize
  File "C:\Users\#########\Desktop\SET22\utils\freecad\FreeCAD-0.20.1-WIN-x64-portable-1\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
[/code]
```
About info (using a portable app):
```
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Portuguese/Brazil (pt_BR)
Installed mods: 
  * BIM 2021.12.0
  * dodo
  * Glass
  * InventorLoader 1.3.0
  * ModernUI (Disabled)
  * sheetmetal 0.2.57
[/code]
```

mado987, 2022-09-23:
Same error :
03:09:19  invalid syntax (BimSetup.py, line 464)
03:09:19  Traceback (most recent call last):
  File "<string>", line 84, in Initialize
  File "C:\Program Files\FreeCAD 0.20\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)

On ->
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: French/Belgium (fr_BE)
Installed mods: 
  * 3DfindIT 1.2.0
  * Assembly3 0.11.3
  * BIM 2021.12.0
  * dodo
  * fasteners 0.4.6
  * parts_library
  * Trails 2022.1.0
[/code]

yorikvanhavre, 2022-09-23:
Committed too fast :sweat_smile: Thanks for the fix!

PR #99 (2022-09-23): fix #98
fix #98
commit 6aa560ba (2022-09-23): fix #98
closed 2022-09-23 by yorikvanhavre