yorikvanhavre/BIM_Workbench #102: Version 0.20 - Internet connection problems - BIM Workbench[Issue]

BloodRaven77, 2022-11-30:
I installed the BIM workbench. After that I went through the setup they recommended. Then when I wanted to go through the tutorial, it says I have to be connected to the internet, but my laptop is always connected to Wi-Fi with internet access.
The message that is displayed in the Report View panel is,
"Unable to access the tutorial. Verify that you are online (This is needed only once)."
[https://user-images.githubusercontent.com/68442233/204740261-b95e99a4-eca9-42f3-a706-18c6d7c0330b.png]

yorikvanhavre, 2022-12-02:
That's weird... Could you try this in the Python console of FreeCAD?
```python
import urllib.request
u=urllib.request.urlopen("https://www.freecadweb.org/wiki/BIM_ingame_tutorial")
print(u.read())
```
It should print a lot of HTML code...

Lord-Buckethead, 2023-05-12:
I had the same issue. I entered this in the console:
```
import urllib.request
u=urllib.request.urlopen("https://www.freecadweb.org/wiki/BIM_ingame_tutorial")
```
Got this back

```
Traceback (most recent call last):
  File "s:\Program Files\FreeCAD 0.20\bin\lib\urllib\request.py", line 1354, in do_open
    h.request(req.get_method(), req.selector, req.data, headers,
  File "s:\Program Files\FreeCAD 0.20\bin\lib\http\client.py", line 1252, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "s:\Program Files\FreeCAD 0.20\bin\lib\http\client.py", line 1298, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "s:\Program Files\FreeCAD 0.20\bin\lib\http\client.py", line 1247, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "s:\Program Files\FreeCAD 0.20\bin\lib\http\client.py", line 1007, in _send_output
    self.send(msg)
  File "s:\Program Files\FreeCAD 0.20\bin\lib\http\client.py", line 947, in send
    self.connect()
  File "s:\Program Files\FreeCAD 0.20\bin\lib\http\client.py", line 1421, in connect
    self.sock = self._context.wrap_socket(self.sock,
  File "s:\Program Files\FreeCAD 0.20\bin\lib\ssl.py", line 500, in wrap_socket
    return self.sslsocket_class._create(
  File "s:\Program Files\FreeCAD 0.20\bin\lib\ssl.py", line 1040, in _create
    self.do_handshake()
  File "s:\Program Files\FreeCAD 0.20\bin\lib\ssl.py", line 1309, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1131)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "s:\Program Files\FreeCAD 0.20\bin\lib\urllib\request.py", line 1354, in do_open
    h.request(req.get_method(), req.selector, req.data, headers,
  File "s:\Program Files\FreeCAD 0.20\bin\lib\http\client.py", line 1252, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "s:\Program Files\FreeCAD 0.20\bin\lib\http\client.py", line 1298, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "s:\Program Files\FreeCAD 0.20\bin\lib\http\client.py", line 1247, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "s:\Program Files\FreeCAD 0.20\bin\lib\http\client.py", line 1007, in _send_output
    self.send(msg)
  File "s:\Program Files\FreeCAD 0.20\bin\lib\http\client.py", line 947, in send
    self.connect()
  File "s:\Program Files\FreeCAD 0.20\bin\lib\http\client.py", line 1421, in connect
    self.sock = self._context.wrap_socket(self.sock,
  File "s:\Program Files\FreeCAD 0.20\bin\lib\ssl.py", line 500, in wrap_socket
    return self.sslsocket_class._create(
  File "s:\Program Files\FreeCAD 0.20\bin\lib\ssl.py", line 1040, in _create
    self.do_handshake()
  File "s:\Program Files\FreeCAD 0.20\bin\lib\ssl.py", line 1309, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1131)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "s:\Program Files\FreeCAD 0.20\bin\lib\urllib\request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
  File "s:\Program Files\FreeCAD 0.20\bin\lib\urllib\request.py", line 525, in open
    response = self._open(req, data)
  File "s:\Program Files\FreeCAD 0.20\bin\lib\urllib\request.py", line 542, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
  File "s:\Program Files\FreeCAD 0.20\bin\lib\urllib\request.py", line 502, in _call_chain
    result = func(*args)
  File "s:\Program Files\FreeCAD 0.20\bin\lib\urllib\request.py", line 1397, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
  File "s:\Program Files\FreeCAD 0.20\bin\lib\urllib\request.py", line 1357, in do_open
    raise URLError(err)
urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1131)>
```

yorikvanhavre, 2023-05-12:
Ok there is apparently something missing in the bundled urllib on windows, but I have no clue how to solve...
We could use http instead of https of course, that would solve the issue, but probably some people will scream that is insecure...
Could you try this?
```python
import urllib.request
u=urllib.request.urlopen("http://www.freecad.org/wiki/BIM_ingame_tutorial")
print(u.read())
```
Maybe we could leave the https by default, but set some preference option to use http for people with problems

Lord-Buckethead, 2023-05-12:
It did not fix the issue:

`>>> import urllib.request
>>> u=urllib.request.urlopen("http://www.freecad.org/wiki/BIM_ingame_tutorial")
Traceback (most recent call last):
  File "S:\Program Files\FreeCAD 0.20\bin\lib\urllib\request.py", line 1354, in do_open
    h.request(req.get_method(), req.selector, req.data, headers,
  File "S:\Program Files\FreeCAD 0.20\bin\lib\http\client.py", line 1252, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "S:\Program Files\FreeCAD 0.20\bin\lib\http\client.py", line 1298, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "S:\Program Files\FreeCAD 0.20\bin\lib\http\client.py", line 1247, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "S:\Program Files\FreeCAD 0.20\bin\lib\http\client.py", line 1007, in _send_output
    self.send(msg)
  File "S:\Program Files\FreeCAD 0.20\bin\lib\http\client.py", line 947, in send
    self.connect()
  File "S:\Program Files\FreeCAD 0.20\bin\lib\http\client.py", line 1421, in connect
    self.sock = self._context.wrap_socket(self.sock,
  File "S:\Program Files\FreeCAD 0.20\bin\lib\ssl.py", line 500, in wrap_socket
    return self.sslsocket_class._create(
  File "S:\Program Files\FreeCAD 0.20\bin\lib\ssl.py", line 1040, in _create
    self.do_handshake()
  File "S:\Program Files\FreeCAD 0.20\bin\lib\ssl.py", line 1309, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1131)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "S:\Program Files\FreeCAD 0.20\bin\lib\urllib\request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
  File "S:\Program Files\FreeCAD 0.20\bin\lib\urllib\request.py", line 531, in open
    response = meth(req, response)
  File "S:\Program Files\FreeCAD 0.20\bin\lib\urllib\request.py", line 640, in http_response
    response = self.parent.error(
  File "S:\Program Files\FreeCAD 0.20\bin\lib\urllib\request.py", line 563, in error
    result = self._call_chain(*args)
  File "S:\Program Files\FreeCAD 0.20\bin\lib\urllib\request.py", line 502, in _call_chain
    result = func(*args)
  File "S:\Program Files\FreeCAD 0.20\bin\lib\urllib\request.py", line 755, in http_error_302
    return self.parent.open(new, timeout=req.timeout)
  File "S:\Program Files\FreeCAD 0.20\bin\lib\urllib\request.py", line 525, in open
    response = self._open(req, data)
  File "S:\Program Files\FreeCAD 0.20\bin\lib\urllib\request.py", line 542, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
  File "S:\Program Files\FreeCAD 0.20\bin\lib\urllib\request.py", line 502, in _call_chain
    result = func(*args)
  File "S:\Program Files\FreeCAD 0.20\bin\lib\urllib\request.py", line 1397, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
  File "S:\Program Files\FreeCAD 0.20\bin\lib\urllib\request.py", line 1357, in do_open
    raise URLError(err)
urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate has expired (_ssl.c:1131)>`

Note, using Firefox it has some issue with the certificate too:

[image]

yorikvanhavre, 2023-05-12:
This is really weird... I have different info!

[image]