yorikvanhavre/BIM_Workbench #103: Ability to `diff` BIM/IFC projects to see what has changed

luzpaz, 2022-12-01:
The possibility to graphically show areas of a project that have been changed in different iterations. I plan to post follow-up graphics when I find them.

yorikvanhavre, 2022-12-02:
There "kind of is" already:
[image]
https://yorik.uncreated.net/blog/2019-009
https://github.com/yorikvanhavre/BIM_Workbench/blob/master/BimDiff.py

I'll leave this open because it would do good to do a new batch of testing, long time I didn't use this tool :)

I say "kind of" because it does a brute-force shape comparison and doesn't consider finer changes in IFC objects. A much better approach, probably, would be to make IFC files more reliable, that's basically the NativeIFC idea: https://github.com/brunopostle/ifcmerge/blob/main/docs/whitepaper.rst

luzpaz, 2022-12-02:
CC @brunopostle (just giving a heads-up)