yorikvanhavre/BIM_Workbench #108: Overlapping Text in Welcome Message [Mac]

nelsonic, 2023-04-04:
Hi! Just following @yorikvanhavre's Getting Started Video: https://youtu.be/Jc12_G5DFF0 📺 
Installed the `BIM` Workbench Addon, restarted `FreeCAD` and selected the `BIM` Workbench: 

[image]

As you can see from this screenshot the welcome message is over the intro image and it looks weird. 

# Info

Device: MacBookPro M1 16"
OS: `13.2.1 (22D68)`
FreeCAD install:
```
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * BIM 2021.12.0
```
Think it might be an issue with the height: 🔍 
https://github.com/yorikvanhavre/BIM_Workbench/blob/d0779e7791acf65c10c55eda93425d005e6a7c3c/dialogWelcome.ui#L22
Happy to try and fix this and create a PR. 🧑‍💻 
But also happy for you to just ignore this as a non-issue that only affects people with a specific screen size ... 💭 

Either way thank you for creating this workbench and tutorials. ❤️

luzpaz, 2023-04-05:
Thanks for the heads up!  
@yorikvanhavre have you seen reports like this before from macOS users ?

nelsonic, 2023-04-05:
FYI: tried on the Windows 11 machine in our workshop and the BIM Workbench Welcome looks fine: 👌 

[image]

Don't know if there are stats on how many people use Mac with FreeCAD. 
But feel free to assign low priority to this issue or even close it.

luzpaz, 2023-04-06:
Weird that macOS displays this.

yorikvanhavre, 2023-04-07:
That's weird! I'll have a look, there might be a simple fix to do in qtdesigner. @nelsonic are you able to resize the dialog by dragging its edges? And if yes does it change anything? Ie, would a bigger or smaller dialog solve the issue?

nelsonic, 2023-04-10:
Not used `qtdesigner` but have some XP with building `C++` GUI apps so could give it a go. 💭 
Reading the manual: https://doc.qt.io/qt-6/qtdesigner-manual.html 👀

yorikvanhavre, 2023-04-11:
Basically the file is this one: https://github.com/yorikvanhavre/BIM_Workbench/blob/master/dialogWelcome.ui
There is no need to compile anything really, you can just open the file in QtDesigner, and try to resize the dialog. 
(You can then replace the file in your BIM workbench folder and simply launch FreeCAD to see the changes)

Indeed I see now that the different labels can overlap each other if the dialog is shrinked. That must be what is happening for you. But I cannot find how to prevent that...

luzpaz, 2023-04-11:
This looks like some sort of layout issue in Qt, right? https://doc.qt.io/qt-5/layout.html

nelsonic, 2023-04-12:
Busy installing `Qt` and got to the following `Setup` screen: 

[image]

Do I need the "`Qt` libraries for clang"?