yorikvanhavre/BIM_Workbench #124: 0.21.1 AppImage: inside Addon Manager, link to NativeIFC addon fails to render

pawsey-kbuckley, 2023-11-29:
0.21.1 AppImage: inside Addon Manager, link to NativeIFC addon fails to render

When you open up the Addon Manager and select the BIM addon, it renders,
or rather it says it is rendering the README.md from

https://github.com/yorikvanhavre/BIM_Workbench/blob/master/README.md

and within that text, there a link to the NativeIFC addon which is

https://github.com/yorikvanhavre/FreeCAD-NativeIFC/tree/main

however, clicking on that link inside the Addon Manager results in a
nessage that says

  Could not load README data from URL https://github.com/yorikvanhavre/FreeCAD-NativeIFC/tree/main

although, within GitHub, the link does take you to the right place.

Wierd!

pawsey-kbuckley, 2023-11-30:
Bit of extra data, in that, what I saw on the console from which I had
invoked the AppImage, were the following lines:

``
"Empty widget item in QVBoxLayout 'verticalLayout'."
[25795:25921:1129/161406.524819:ERROR:nss_util.cc(283)] After loading Root Certs, loaded==false: NSS error code: -8018
Release of profile requested but WebEnginePage still not deleted. Expect troubles !
Release of profile requested but WebEnginePage still not deleted. Expect troubles !
Release of profile requested but WebEnginePage still not deleted. Expect troubles !
Release of profile requested but WebEnginePage still not deleted. Expect troubles !
```

yorikvanhavre, 2023-12-01:
Is the addon manager able to open links in readme's at all?

pawsey-kbuckley, 2023-12-05:
Don't really have a definitive answer for you: apologies.

If I click on a link within the BIM "readme" I do get moved around with the readme,
although it's not clear that I'm moved to the correct target/anchor, but the external
link, to the IFC URI still fails, however,

if I open up the BCF Plugin, and click on the FreeCAD link in there, I do get taken to 
a page outside of the BCF Plugin README.

HTH

chennes, 2023-12-05:
It's supposed to launch those links in your system web browser, it's possible something got broken there.

yorikvanhavre, 2023-12-11:
It seems to open the link briefly in the same window (inside the addon manager). What I can see so far:
- non-github links open in the desktop browser
- github links keep opening in the addon manager, but they get parsed with the markdown converter
I think the best solution is to make sure only links ending with .md get parsed with the markdown converter, the others should be opened in the defautl browser too... I'll have a look at it later