yorikvanhavre/BIM_Workbench #130: IFC preflight reports doors and windows as not being a part of any `IfcBuildingStorey`

p-himik, 2024-01-02:
When running the IFC preflight check on a model with a door that has a host inside a proper level, it still results in the "Are all BIM objects part of a level?" check marked as "Failed".

It seems that a workaround is to add every door/window to a level explicitly, even when they have a host that's added to a level already.

A minimal model is attached.
[ifc_door_preflight_issue.zip](https://github.com/yorikvanhavre/BIM_Workbench/files/13810030/ifc_door_preflight_issue.zip)

p-himik, 2024-01-02:
I just realized that the same door/window can have multiple hosts that can belong to different levels. Not sure how reasonable or common it is. If it's possible to attach a door/window automatically to the same level that its sole host belongs to during IFC preflight check and export, perhaps it makes sense to have a separate check specifically for either a sole host or an explicit attachment to a level for every door/window.

yorikvanhavre, 2024-01-08:
I think IFC indeed requires for a door or window to be specifically related to a containing structure (Storey, building...). Being used by a wall that is part of that structure is not enough.

Now indeed in FreeCAD that's a detail you usually don't need to care about. I think if a door/window has any object as its Host that is in a building structure, we should indeed consider it automatically part of that structure too...

commit a0e7bb8b (2024-01-08): Fixed preflight test - fixes #130
closed 2024-01-08 by yorikvanhavre