yorikvanhavre/BIM_Workbench #144: import ifc_import error

tibohu, 2024-09-30:
Hi there! I tried to do"import ifc_import" via the python console with both 0.21.2 and 1.0.0RC2.
As described on https://github.com/yorikvanhavre/FreeCAD-NativeIFC/blob/main/doc/README.md#import-an-ifc-file
(For both versions, the required addon is installed ifcopenshell as well. The ifcopenshell import is successful.)
But both cases are unsuccessful at "import ifc_import".

yorikvanhavre, 2024-09-30:
Note that in v1.0RC nativeIFC is fully integrated into BIM, you don't need to install anything else anymore. But you need to do `from nativeifc import ifc_import` instead of just `import ifc_import`.

If this still gives errors, can you please copy here the text of the error message?

tibohu, 2024-09-30:
Dear Yorik! Thank you! This way it works well.
Is there a more up to date version of FreeCAD-NativeIFC tutorial with python snippets than the one I linked above?

yorikvanhavre, 2024-09-30:
No unfortunately :/ But that's on my todo list. I'll reconvert this issue so we don't keep track

closed 2024-09-30 by yorikvanhavre